asus-linux-drivers/asus-numberpad-driver #231: [Feature Request] Add support for eopkg (Solus)

elia-414, 2025-02-18:
### Short Description

Please add support for eopkg package manager on Solus, both Wayland and xorg.

ldrahnik, 2025-02-19:
@elia-414 Hello, thank you for creating an issue. Please, try the latest master branch and let me know.

elia-414, 2025-03-05:
Hello, here is the error, maybe there's different package to refer to? System error. Program terminated.
Repo item python3-virtualenv not found

ldrahnik, 2025-03-05:
@elia-414 Please, try the latest master now.

elia-414, 2025-03-05:
Unfortunately pkg-config is not in repo (maybe under another name? I tried searching, but didn't manage to). And modules are located in different location: tee: /etc/modules-load.d/uinput-asus-numberpad-driver.conf: No such file or directory
tee: /etc/modules-load.d/i2c-dev-asus-numberpad-driver.conf: No such file or directory
Something went wrong when adding uinput module to auto loaded modules

ldrahnik, 2025-03-05:
@elia-414 `eopkg search pkg-config` returns something?

ldrahnik, 2025-03-05:
@elia-414 Where are located modules?

elia-414, 2025-03-06:
I apologize for bothering you a lot with this. /etc/modules-load.d needs to be created manually. The only thing left is pkg-config. In Solus repo there is a package "python-pkgconfig" - "A Python interface to the pkg-config command line tool" - is it the right one? For now installation went smoothly apart from pkg-config part.

ldrahnik, 2025-03-06:
@elia-414 No, what the driver requires is the system-wide `pkg-config` utility, which is usually just called `pkg-config`. Try to comment out `pkg-config` and try to use NumberPad, post the error log here.

ldrahnik, 2025-03-06:
@elia-414 I recommend ask on distribution forum how install this package.

elia-414, 2025-03-09:
So pkg-config is known as `pkgconf` in Solus repo, could you please change it? I have manually changed it in install.sh and it worked. For Solus development tools should be also installed:
`# eopkg install -c system.devel`

It works on my ASUS UX425EA, thank you for your work and thank you for helping. I am sorry if I was bothering too much!

ldrahnik, 2025-03-10:
@elia-414 Great news (y) Np, I would use `pkg-search`, I would ask on forum etc. aswell. I have never installed Solus distro. I pushed creating directory `/etc/modules-load.d` and fix of `pkg-conf` package. Let me know if anything else is needed.

closed 2025-03-10 by ldrahnik