asus-linux-drivers/asus-numberpad-driver #232: [Bug]: asus-numpad driver slows down layout change.

RomanSyunyaev, 2025-03-04:
### Describe the bug

I have multilanguage keyboard configuration on gnome\Debian, layout is changed by Caps Lock . Driver installation results in the ~2 seconds delay between Caps Lock button press and actual switch of layout.
Uninstallation of the driver immediately fixes the problem.

### Expected behavior

Driver should not affect language layout change. Layout is expected to change immediately after the button press.

### Relevant log output

```shell
None of the logs seems to be relevant. But I am attaching an example just in case.

$journalctl -f | grep asus-numberpad-driver
Hint: You are currently not seeing messages from other users and the system.
      Users in groups 'adm', 'systemd-journal' can see all messages.
      Pass -q to turn off this notice.
Mar 04 22:45:50 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Touched top_right_icon area (numlock) in time: 1741128350.093705
Mar 04 22:45:50 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Slide from top_right_icon exceeded the activation threshold for x and y.
Mar 04 22:45:50 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Activation radius 1200.00 (top left corner is 0)
Mar 04 22:45:50 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: System numlock activated
Mar 04 22:45:50 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Numpad activated
Mar 04 22:45:50 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Setting up for config file key: "enabled" with value: "True"
Mar 04 22:45:50 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
Mar 04 22:46:00 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Touched top_right_icon area (numlock) in time: 1741128360.6389334
Mar 04 22:46:00 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Slide from top_right_icon exceeded the activation threshold for x and y.
Mar 04 22:46:00 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Activation radius 1200.00 (top left corner is 0)
Mar 04 22:46:00 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: System numlock deactivated
Mar 04 22:46:00 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Numpad deactivated
Mar 04 22:46:00 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Setting up for config file key: "enabled" with value: "False"
Mar 04 22:46:00 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
Mar 04 22:46:07 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Touched top_right_icon area (numlock) in time: 1741128367.235275
Mar 04 22:46:07 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Slide from top_right_icon exceeded the activation threshold for x and y.
Mar 04 22:46:07 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Activation radius 1200.00 (top left corner is 0)
Mar 04 22:46:07 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: System numlock activated
Mar 04 22:46:07 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Numpad activated
Mar 04 22:46:07 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Setting up for config file key: "enabled" with value: "True"
Mar 04 22:46:07 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
566Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_5:6
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_5:6
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
266*Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_2:3
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_2:3
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: [KEY_LEFTSHIFT:42, KEY_8:9]
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: [KEY_LEFTSHIFT:42, KEY_8:9]
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:08 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
[... 133 lines cut ...]
Mar 04 22:46:14 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:14 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_9:10
Mar 04 22:46:14 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:14 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_9:10
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_5:6
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_5:6
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Pressed numpad key
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Unpressed numpad key
Mar 04 22:46:15 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: KEY_6:7
Mar 04 22:46:16 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Touched top_right_icon area (numlock) in time: 1741128376.4312203
Mar 04 22:46:16 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Slide from top_right_icon exceeded the activation threshold for x and y.
Mar 04 22:46:16 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Activation radius 1200.00 (top left corner is 0)
Mar 04 22:46:16 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: System numlock deactivated
Mar 04 22:46:16 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Numpad deactivated
Mar 04 22:46:16 ras /usr/share/asus-numberpad-driver/numberpad.py[1900]: Setting up for config file key: "enabled" with value: "False"
```

### Desktop

- Program version [6.5.0]
- Distro [Debian]
- Kernel [6.12.12+bpo-amd64]

RomanSyunyaev, 2025-03-04:
Very similar issue was actually described in a number of posts years ago, and seems to be something gnome-related.
For example:
https://bbs.archlinux.org/viewtopic.php?id=153833
I am not sure if anyone was able to find out what exactly causes the delay though.

ldrahnik, 2025-03-13:
@RomanSyunyaev Hello, at first thank you for creating an issue. I confirm that the delay can be caused by `sleep(0.5)` in the listening thread. Could you test this version that avoids to using `sleep`?

```
#!/bin/bash
# Duration in seconds (change this as needed)
DURATION=1
# Find touchpad device ID
DEVICE_ID=$(libinput list-devices | grep -i -B 1 'touchpad' | grep 'Device:' | awk -F 'Device: ' '{print $2}')
# Disable touchpad
echo "Disabling touchpad: $DEVICE_ID"
qdbus org.kde.KWin /org/kde/KWin/InputDevice/touchpad org.kde.KWin.InputDevice.setEnabled false
# Wait for the specified duration
sleep $DURATION
# Enable touchpad
echo "Re-enabling touchpad: $DEVICE_ID"
qdbus org.kde.KWin /org/kde/KWin/InputDevice/touchpad org.kde.KWin.InputDevice.setEnabled true
```

closed 2025-03-14 by ldrahnik