asus-linux-drivers/asus-numberpad-driver #234: [Bug]: NixOS flake installation

veseckytomas, 2025-03-22:
### Describe the bug

Installing using flake.nix and configuration snipped fails to run.
Configuration: https://gitlab.com/vesecky.tomas/thompson-nix/-/tree/asus-numberpad
Hardware: ASUS Zenbook 14 UX3405CA-OLED232X (so default layout up5401ea is correct)

### Expected behavior

Numberpad driver installed and running.

### Relevant log output

```shell
Mar 22 21:59:08 TomasVesa-NTB-NixOS /nix/store/78f9i368mmd5fjf7zfa8pz0sl6jcvghn-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[1280]: X11 detected but not connected succesfully to the display None. Exiting
Mar 22 21:59:08 TomasVesa-NTB-NixOS numberpad.py[1280]: 2025-03-22 21:59:08,384 ERROR X11 detected but not connected succesfully to the display None. Exiting
Mar 22 21:59:08 TomasVesa-NTB-NixOS systemd[1]: asus-numberpad-driver.service: Main process exited, code=exited, status=1/FAILURE
```

### Desktop

- Program version [410c32e]
- Distro [NixOS]
- Kernel [Linux 6.13.7-zen1]

veseckytomas, 2025-03-22:
@scientiac I would appreciate your input to this issue.

ldrahnik, 2025-03-22:
@veseckytomas Hello, I have added the required env variable `DISPLAY` to the config. Check in your terminal using `$ echo $DISPLAY` and change the default value `:0` if necessary. Let me know.

scientiac, 2025-03-23:
I had not really tested it in X11 environment, but previously the service run as root, now it is run by user (--user) and it now works with @ldrahnik adding the default `DISPLAY` env var.  I am currently using niri compositor which by default doesn't have x-wayland buillt-in. But upon running the service with x11 and running an external x-wayland server ([xwayland-satellite](https://github.com/Supreeeme/xwayland-satellite)) it worked.

```nix
  services.asus-numberpad-driver = {
    enable = true;
    wayland = false;

    display = ":0";
    # waylandDisplay = "wayland-1";
    config = { "default_backlight_level" = "0x41"; };
  };
```
A problem is that the program tries to write on the config which NixOS doesn't allow. But it works regardless.
```
Mar 23 11:46:55 einstein numberpad.py[2710]: 2025-03-23 11:46:55,534 INFO Slide from top_right_icon exceeded the activation threshold for x and y.
Mar 23 11:46:55 einstein numberpad.py[2710]: 2025-03-23 11:46:55,534 INFO Activation radius 1200.00 (top left corner is 0)
Mar 23 11:46:55 einstein /nix/store/b4hxxhfmrbklk93j525kir4z0d5kimbk-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[2710]: System numlock deactivated
Mar 23 11:46:55 einstein numberpad.py[2710]: 2025-03-23 11:46:55,535 INFO System numlock deactivated
Mar 23 11:46:55 einstein numberpad.py[2710]: 2025-03-23 11:46:55,535 INFO Numpad deactivated
Mar 23 11:46:55 einstein /nix/store/b4hxxhfmrbklk93j525kir4z0d5kimbk-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[2710]: Numpad deactivated
Mar 23 11:46:55 einstein /nix/store/b4hxxhfmrbklk93j525kir4z0d5kimbk-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[2710]: Setting up for config file key: "enabled" with value: "False"
Mar 23 11:46:55 einstein numberpad.py[2710]: 2025-03-23 11:46:55,536 INFO Setting up for config file key: "enabled" with value: "False"
Mar 23 11:46:55 einstein /nix/store/b4hxxhfmrbklk93j525kir4z0d5kimbk-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[2710]: Error during writting to config file: "/etc/asus-numberpad-driver/numberpad_dev"
Mar 23 11:46:55 einstein numberpad.py[2710]: 2025-03-23 11:46:55,537 ERROR Error during writting to config file: "/etc/asus-numberpad-driver/numberpad_dev"
```

I had to restart my system after enabling the service on nix. Or had to start it manually once using systemd to fully set it up, but now it basically works x11.

I haven't yet made `xkbcommon` to be `<1.1` but it does works as is, so I am not touching it.

I installed gnome to test it and enabling with X11 and checking on both wayland and x11 sessions it just worked without needing to change the config which was surprising. Where previously when I had to switch between these two to work on gnome and niri respectively:

```
    waylandDisplay = "wayland-0";
    waylandDisplay = "wayland-1";
```

It's a very simple change to fix the issue. I'll be sending a pull request shortly after researching if I can make it able to work on both `wayland-1` and `wayland-0` without switching configs and rebuilding.

closed 2025-03-23 by ldrahnik

ldrahnik, 2025-03-23:
> Mar 23 11:46:55 einstein numberpad.py[2710]: 2025-03-23 11:46:55,537 ERROR Error during writting to config file: "/etc/asus-numberpad-driver/numberpad_dev"

@scientiac @veseckytomas That should not be a problem. It only saves the current states: `enabled` and `idled` to the config so other associated whatever could read these states outside of driver scope ..

veseckytomas, 2025-03-23:
Issue still persists with different error.

What I tried
1. nix flake update to pull changes
2. update my configuration (see https://gitlab.com/vesecky.tomas/thompson-nix/-/tree/asus-numberpad)

output of journalctl --user -xeu asus-numberpad-driver.service:
```
Mar 23 10:24:45 TomasVesa-NTB-NixOS /nix/store/6b0i6m8vxr4q17wy3finqgyv4cfp3wq9-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[26864]: X11 has xauth file in /tmp folder with filename changed each boot, currently /tmp/xauth_wKIusJ
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]: 2025-03-23 10:24:45,404 INFO X11 has xauth file in /tmp folder with filename changed each boot, currently /tmp/xauth_wKIusJ
Mar 23 10:24:45 TomasVesa-NTB-NixOS /nix/store/6b0i6m8vxr4q17wy3finqgyv4cfp3wq9-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[26864]: X11 detected and connected succesfully to the display :0
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]: 2025-03-23 10:24:45,422 INFO X11 detected and connected succesfully to the display :0
Mar 23 10:24:45 TomasVesa-NTB-NixOS /nix/store/6b0i6m8vxr4q17wy3finqgyv4cfp3wq9-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py[26864]: Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]: 2025-03-23 10:24:45,424 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]: Traceback (most recent call last):
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]:   File "/nix/store/6b0i6m8vxr4q17wy3finqgyv4cfp3wq9-python3.11-asus-numberpad-driver-6.5.0/share/asus-numberpad-driver/numberpad.py", line 925, in <module>
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]:     with open('/dev/input/event' + str(keyboard), 'rb') as fd_k:
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 23 10:24:45 TomasVesa-NTB-NixOS numberpad.py[26864]: PermissionError: [Errno 13] Permission denied: '/dev/input/event0'
Mar 23 10:24:45 TomasVesa-NTB-NixOS systemd[2457]: asus-numberpad-driver.service: Main process exited, code=exited, status=1/FAILURE
```

scientiac, 2025-03-23:
@veseckytomas 
Add your user to these groups and try again:

extraGroups = [ "i2c" "input" "uinput" ];

veseckytomas, 2025-03-23:
Group input did the trick. Thank you so much for help.

ldrahnik, 2025-04-08:
@scientiac Reverted running as user service https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/b1891bd7bc250719fbf50f8e2d676330a7ca15cc / https://github.com/asus-linux-drivers/asus-dialpad-driver/commit/d68f8c788120093c78a09fad33818b1121a6d392. An idea about driver-specific user mentioned here https://github.com/asus-linux-drivers/asus-dialpad-driver/issues/10#issuecomment-2786691039

PR #235 (2025-03-23): [#234] fix: Installation on X11 on NixOS, add: ignoreWaylandDisplayEnv config option
1. Runs the service as user, not root.
2. Add an option `ignoreWaylandDisplayEnv = <bool>;` defaulting to `false`. (Not using `WAYLAND_DISPLAY` env var on the systemd service fixed my issue of having to rebuild the config when switching between gnome and niri.)
3. Clarify about added configuration on README and some more improvements.

Didn't completely remove the WAYLAND_DISPLAY option to preserve compatibility and avoid regressions.

Fixes #234