asus-linux-drivers/asus-numberpad-driver #237: [Bug]: UN4501QAB Numberpad randomly is turning on and off

xJustDoNothing, 2025-03-23:
### Describe the bug

After installation my numberpad is turning randomly on and off. I can turn it off manually by pressing the button in the top left but it will turn on as soon as i start typing on my keyboard again it turns on again. Is there a way to deactivate this? 

### Expected behavior

It should be turned off till I turn it back on again

### Relevant log output

```shell

```

### Desktop

- Program version [e.g. 57c742a]
- Distro [Arch Linux]
- Kernel [6.13.7-arch1-1]
-  WinowManager [Hyprland]

ldrahnik, 2025-03-23:
@xJustDoNothing Hello, could you post here your config?

xJustDoNothing, 2025-03-23:
```shell
[main]
numpad_disables_sys_numlock = 1
disable_due_inactivity_time = 0
touchpad_disables_numpad = 1
key_repetitions = 0
multitouch = 0
one_touch_key_rotation = 0
activation_time = 1
sys_numlock_enables_numpad = 1
top_left_icon_activation_time = 1
top_left_icon_slide_func_activation_radius = 1200
top_left_icon_slide_func_activates_numpad = 1
top_right_icon_slide_func_activation_radius = 1200
enabled_touchpad_pointer = 3
press_key_when_is_done_untouch = 1
enabled = 1
top_left_icon_brightness_func_max_min_only = 0
default_backlight_level = 0x01
brightness = 0x48
top_left_icon_brightness_func_disabled = 0
distance_to_move_only_pointer = 0
idled = 0
idle_brightness = 30
idle_enabled = 0
idle_time = 10
top_left_icon_slide_func_disabled = 0

```

xJustDoNothing, 2025-03-23:
@ldrahnik I put it in my initial report

ldrahnik, 2025-03-23:
@xJustDoNothing Is that latest master branch? Try `git clone .. ` and install latest version if not.

xJustDoNothing, 2025-03-23:
Yes, I cloned the repo today

ldrahnik, 2025-03-23:
@xJustDoNothing Try to change in config: `numpad_disables_sys_numlock` to `0`, `enabled = 1` in config change to `0`, should help. Also please post here output of `/proc/bus/input/devices`.

ldrahnik, 2025-03-23:
@xJustDoNothing The problem started since was driver installed? Whole time?

ldrahnik, 2025-03-23:
@xJustDoNothing Also please post here log from installation, is located here `/var/log/asus-nunberpad-driver/*`.

xJustDoNothing, 2025-03-23:
I edited the config, but even after a restart nothing changed, do i need to source the config or smth to appy the changes and yes the problem started only since i installed the driver for the first time.

### First install log 
'''shell
warning: ibus-1.5.31-2 is up to date -- skipping
warning: libevdev-1.13.3-1 is up to date -- skipping
warning: curl-8.12.1-1 is up to date -- skipping
warning: python-3.13.2-1 is up to date -- skipping
warning: libxml2-2.13.6-3 is up to date -- skipping
warning: libxkbcommon-1.8.1-1 is up to date -- skipping
warning: gcc-14.2.1+r753+g1cd744a6828f-1 is up to date -- skipping
warning: pkgconf-2.4.3-1 is up to date -- skipping
warning: systemd-257.4-1 is up to date -- skipping
resolving dependencies...
looking for conflicting packages...

Package (5)              New Version  Net Change  Download Size

extra/python-distlib     0.3.9-2        1.29 MiB       0.24 MiB
extra/xorg-xrandr        1.5.3-1        0.07 MiB       0.04 MiB
extra/i2c-tools          4.4-2          0.32 MiB       0.09 MiB
extra/python-virtualenv  20.28.0-1      5.19 MiB       3.36 MiB
extra/xorg-xinput        1.6.4-2        0.06 MiB       0.03 MiB

Total Download Size:   3.75 MiB
Total Installed Size:  6.94 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 python-virtualenv-20.28.0-1-any downloading...
 python-distlib-0.3.9-2-any downloading...
 i2c-tools-4.4-2-x86_64 downloading...
 xorg-xrandr-1.5.3-1-x86_64 downloading...
 xorg-xinput-1.6.4-2-x86_64 downloading...
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
checking available disk space...
:: Processing package changes...
installing xorg-xrandr...
installing xorg-xinput...
installing i2c-tools...
Optional dependencies for i2c-tools
    read-edid: for decode-edid script
    python: for smbus module [installed]
installing python-distlib...
installing python-virtualenv...
:: Running post-transaction hooks...
(1/3) Creating system user accounts...
Creating group 'i2c' with GID 960.
(2/3) Reloading device manager configuration...
(3/3) Arming ConditionNeedsUpdate...
warning: wayland-1.23.1-1 is up to date -- skipping
 there is nothing to do

groupadd: group 'input' already exists
groupadd: group 'i2c' already exists
groupadd: group 'uinput' already exists
Added groups input, i2c, uinput, numberpad to current user
uinput module loaded
i2c-dev module loaded
uinput module added to auto loaded modules
Udev rules reloaded and triggered

Testing interface i2c-0: failed
Testing interface i2c-1: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-1

Default config will be autocreated during the first run and available for futher modifications here:
[... 390 lines cut ...]
B: EV=21
B: SW=4

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="Asus WMI accel tablet mode"
P: Phys=
S: Sysfs=/devices/virtual/input/input34
U: Uniq=
H: Handlers=kbd event20 
B: PROP=0
B: EV=23
B: KEY=200000 0 0
B: SW=2

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="ASUE140D:00 04F3:31B9 NumberPad"
P: Phys=
S: Sysfs=/devices/virtual/input/input35
U: Uniq=
H: Handlers=kbd event21 
B: PROP=0
B: EV=3
B: KEY=70000 0 1000 20 231440378444ffc

'''

ldrahnik, 2025-03-23:
@xJustDoNothing First deactivate NumberPad, are you able do that? Then change the config, both values. Then enable, it should help.

ldrahnik, 2025-03-23:
@xJustDoNothing Try this for deactivating: ```sed -i "s/enabled = 1/enabled = 0/g" /usr/share/asus-numberpad-driver/numberpad_dev```

ldrahnik, 2025-03-23:
@xJustDoNothing When this happens, is your NumLock indicator toggling?

xJustDoNothing, 2025-03-23:
@ldrahnik I can't deactivate it and i can't see the NumLock indicator toggling when i do the commands u said. I mean i know from windows that the only difference of the numpad activating or deactivating is that the numbers on the pad are illuminated

xJustDoNothing, 2025-03-23:
@ldrahnik oh nvm just noticed it turns off for a split second and then instantly turns back on again if i do the sed -i command
-

xJustDoNothing, 2025-03-23:
Omg I think i fixed it, I played around in the config and turned the `sys_numlock_enables_numpad ` to `0 `as well as the 
`numpad_disables_sys_numlock`to '0' this seems to have fixed the issue for now

ldrahnik, 2025-04-08:
@xJustDoNothing Have you encountered this problem since then?

ldrahnik, 2025-04-11:
Can be reopened.