asus-linux-drivers/asus-numberpad-driver #239: [Bug]: numberpad works, but no light (Ubuntu 24.10)

alainlhostis, 2025-04-04:
### Describe the bug

I ran the script provided with bash install.sh
I chose all options with 'yes'
Numpad works by 1 sec touch on the top right corner, but light remains off

### Expected behavior

there should be some lighting as in the photos

### Relevant log output

```shell

```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2025-04-07:
@alainlhostis Hello, your laptop is? Does it support NumberPad with Windows. In čase yes, please provider here a logs as is described during creating issue.

ldrahnik, 2025-04-08:
Can be reopened.

closed 2025-04-08 by ldrahnik

alainlhostis, 2025-04-16:
It really seems this model does NOT have a NumberPad, unfortunately: https://www.comparez-malin.fr/informatique/pc-portable/asus-zenbook-s-14-ux5406sa-drpv142w.html?utm_source=chatgpt.com