asus-linux-drivers/asus-numberpad-driver #241: [Bug]: Backlight was not activated

alainlhostis, 2025-04-15:
### Describe the bug

Backlight was not activated
My laptop is ASUS Zenbook S 14 UX5406SA_UX5406SA
Follwed instructions given in readme as:
- activate the numpad by one second press right top 
- `bash install_device_check.sh
[sudo] Mot de passe de dddddddddd : 
Testing interface i2c-0: failed
Testing interface i2c-1: failed
Testing interface i2c-2: failed
Testing interface i2c-3: success (adr 0x15)
The detection was successful. Touchpad with NumberPad found: i2c-3
`
- From this was able to adapt the command as follows:
`i2ctransfer -f -y i2c-3 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x24 0xad
Error: I2C bus name doesn't match any bus present!
`
- but received this error 

And still the backlight is not activated

### Expected behavior

Light should show up

### Relevant log output

```shell

```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

ldrahnik, 2025-04-16:
@alainlhostis Hello, could you post here website for your product with information that your laptop has NumberPad?

alainlhostis, 2025-04-16:
The product: [https://www.asus.com/fr/laptops/for-home/zenbook/asus-zenbook-s-14-ux5406/](https://www.asus.com/fr/laptops/for-home/zenbook/asus-zenbook-s-14-ux5406/)
finally not sure it has a number pad...

alainlhostis, 2025-04-16:
effectively the instruction:
`i2ctransfer -f -y 3 w13@0x15 0x05 0x00 0x3d 0x03 0x06 0x00 0x07 0x00 0x0d 0x14 0x03 0x24 0xad`
does not generate error anymore, but still no backlight on the pad

alainlhostis, 2025-04-16:
It really seems this model does NOT have a NumberPad, unfortunately: https://www.comparez-malin.fr/informatique/pc-portable/asus-zenbook-s-14-ux5406sa-drpv142w.html?utm_source=chatgpt.com

closed 2025-04-16 by ldrahnik