asus-linux-drivers/asus-numberpad-driver #242: [Bug]: Asterix key error

fmagnucz, 2025-04-19:
### Describe the bug

I have a new ASUS Zenbook 14 UX3405CA. I can install your numpad driver with the default layout, and it almost works propoerly. Everything is ok, but I press the asterix (*) key, the numpad turn off immediately. :-(

### Expected behavior

When I press the asterix key the numpad don't turn off and read the write the asterix to the active program (calculator, LibreOffice writer, calc, etc.)

### Relevant log output

```shell

```

### Desktop

- Program version: v6.5.0
- Distro: Fedora 41, Ubuntu 25.04, EndeavourOS_Mercury-Neo-2025.03.19
- Kernel: 6.13, 6.14

ldrahnik, 2025-04-20:
@fmagnucz Hello, could you attach logs, please.

ldrahnik, 2025-04-21:
Can be reopened.

fmagnucz, 2025-04-26:
Yes, but I don't know how. :-/

ldrahnik, 2025-04-27:
@fmagnucz 
>Relevant log output
Please copy and paste any relevant log output. The install and uninstall logs are here /var/log/asus-numberpad-driver/*.log. The runtime logs can be found using journalctl -f | grep asus-numberpad-driver or journalctl -f --user-unit asus_numberpad_driver@$USER.service. This will be automatically formatted into code, so no need for backticks.

Or just stop systemd service `sudo systemctl stop --user asus_numberpad..` and run `python3 numberpad.py in the downloaded repository

ldrahnik, 2025-05-02:
Can be reopened.

fmagnucz, 2025-08-01:
Hi,

Here is the log of the following actions.
1. turn on the number pad.
2. press the asterix key.

Aug 01 22:42:18 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Touched top_right_icon area (numlock) in time: 1754080938.840518
Aug 01 22:42:19 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: The numpad numlock was pressed longer than the activation time and will be activated: 1.0025641918182373
Aug 01 22:42:19 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Activation time: 1.0
Aug 01 22:42:19 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: System numlock activated
Aug 01 22:42:19 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Numpad activated
Aug 01 22:42:19 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Setting up for config file key: "enabled" with value: "True"
Aug 01 22:42:19 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Pressed numpad key
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: [KEY_54:84, KEY_SLASH:53]
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Clean up started
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: System numlock deactivated
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Setting up for config file key: "idled" with value: "False"
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Cancelled numpad idle
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Cancelled idle
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Setting up for config file key: "enabled" with value: "False"
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Numpad deactivated
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Clean up finished
Aug 01 22:42:24 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[5986]: Exiting
Aug 01 22:42:25 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[6083]: Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
Aug 01 22:42:25 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[6083]: Set keyboard 2 from H: Handlers=sysrq kbd leds event2
Aug 01 22:42:25 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[6083]: Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
Aug 01 22:42:25 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[6083]: Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUP1415:00/0018:093A:300C.0001/input/input7
Aug 01 22:42:25 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[6083]: Set touchpad id 4 from H: Handlers=mouse1 event4
Aug 01 22:42:25 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[6083]: Touchpad min-max: x 0-3996, y 0-2242
Aug 01 22:42:25 Krisztina-Notebook /usr/share/asus-numberpad-driver/numberpad.py[6083]: Numpad min-max: x 200-3796, y 200-2162

fmagnucz, 2025-08-01:
Actual operating system: Fedora release 42 (Adams)

$ uname -srvpio
Linux 6.15.6-200.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Jul 10 15:22:32 UTC 2025 unknown unknown GNU/Linux

fmagnucz, 2025-08-01:
Here is the log of all keypressing.

[asus-numberpad-driver-6.5.1.log](https://github.com/user-attachments/files/21554770/asus-numberpad-driver-6.5.1.log)

fmagnucz, 2025-08-01:
[asus-numberpad-driver-6.5.1_events.log](https://github.com/user-attachments/files/21554949/asus-numberpad-driver-6.5.1_events.log)

ldrahnik, 2025-08-04:
@fmagnucz I need to see your mappings, e.g. mine:

```
2025-08-04 19:05:38,572 DEBUG {'Num_Lock': KEY_NUMLOCK:69, '0': KEY_0:11, '1': KEY_1:2, '2': KEY_2:3, '3': KEY_3:4, '4': KEY_4:5, '5': KEY_5:6, '6': KEY_6:7, '7': KEY_7:8, '8': KEY_8:9, '9': KEY_9:10, 'a': KEY_A:30, 'b': KEY_B:48, 'c': KEY_C:46, 'd': KEY_D:32, 'e': KEY_E:18, 'f': KEY_F:33, 'Shift_L': KEY_LEFTSHIFT:42, 'Control_L': KEY_LEFTCTRL:29, 'u': KEY_U:22, 'space': KEY_SPACE:57, 'slash': KEY_SLASH:53, 'BackSpace': KEY_BACKSPACE:14, 'asterisk': [KEY_LEFTSHIFT:42, KEY_8:9], 'minus': KEY_MINUS:12, 'percent': [KEY_LEFTSHIFT:42, KEY_5:6], 'period': KEY_DOT:52, 'Return': KEY_ENTER:28, 'plus': [KEY_LEFTSHIFT:42, KEY_EQUAL:13], 'equal': KEY_EQUAL:13}
```

ldrahnik, 2025-08-06:
@fmagnucz It does not look like a default Hungarian keyboard layout. It is manually customized? Or what should I choose? 

[image]

fmagnucz, 2025-08-17:
Hi @pvojnisek !

I have returned from holiday. I will test your fixed version tomorrow evening and I will reply here and your pull request as well.

fmagnucz, 2025-08-20:
Hi @pvojnisek!

I have tested your fix and it works properly.

@ldrahnik , if you agree with the @pvojnisek's fix you can merge his solution into your code.

ldrahnik, 2025-08-24:
@fmagnucz @pvojnisek Closing because was merged https://github.com/asus-linux-drivers/asus-numberpad-driver/pull/249 to the master branch. Can be reopened.