asus-linux-drivers/asus-numberpad-driver #243: [Bug]: Numpad is toggling twice in most situation on Fedora 42 and `up5401ea` model

franckgaga, 2025-05-15:
### Describe the bug

When I hold the numpad activation button, the toggling occurs two times most the of time, making it unusable. Here's a video:

https://photos.app.goo.gl/buoSDQTuTuFgYkLD6

I used it on Fedora 41 in the past without any issues. 

### Expected behavior

It should toggle only once at each long press.

### Relevant log output

```shell
mai 14 20:45:22 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2320]: check_config_values_changes: detected external change of config file -> loading changes
mai 14 20:45:22 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2318]: Setting up for config file key: "enabled" with value: "False"
mai 14 20:45:22 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2318]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
mai 14 20:45:22 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2320]: Setting up for config file key: "enabled" with value: "False"
mai 14 20:45:22 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2318]: check_config_values_changes: detected internal change of config file -> do nothing -> would be deadlock
mai 14 20:45:23 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2318]: Numpad deactivated
mai 14 20:45:23 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2320]: Numpad deactivated
mai 14 20:45:23 asus-zenbook /usr/share/asus-numberpad-driver/numberpad.py[2320]: check_config_values_changes: detected external change of config file -> loading changes
```

### Desktop

- v6.5.0
- Fedora 42
- 6.14.5-300.fc42

ldrahnik, 2025-05-15:
@franckgaga Hello, in brackets you can see two different process numbers. Try figure out why do you have installed NumberPad driver twice in your system.

franckgaga, 2025-05-15:
I noticed that one process is `francis.service ()` and the other is `francisgagnon.service ()`. I recently refreshed my OS and restored my home directory settings using the backup application "Deja-Dup" (my last username was `francis`, my new username is `francisgagnon`. That's probably the cause. Is there any folder that I could delete in my home directory that I could delete to hopefully solve the issue ?

closed 2025-05-15 by franckgaga