asus-linux-drivers/asus-numberpad-driver #244: suggestion: make it wayland environment agnostic

scientiac, 2025-06-14:
I have both GNOME and Niri installed and switch between the two, 

But the existance of :
```
Environment="WAYLAND_DISPLAY=$WAYLAND_DISPLAY"
```
in the systemd service I can only use the program on one environment.

Niri uses `wayland-1` and GNOME uses `wayland-0`.

But removing the line for WAYLAND_DISPLAY environment from the systemd service will make it be able to work on both the environments.

ldrahnik, 2025-06-15:
@scientiac Hello, thank you! Could you please add what variable `display_wayland_var` below contains on Niri? It contains `wayland-1` or function `display_wayland.connect()` knows how to connect without that? Do you have on Niri `wayland-1` and not used `wayland-0`?

```
display_wayland_var = os.environ.get('WAYLAND_DISPLAY')
```

```
display_wayland = Display(display_wayland_var)
        display_wayland.connect()
```

ldrahnik, 2025-06-15:
@scientiac I would be interested in comparing a dump of all env variables, for example, using:

```
for key, value in os.environ.items():
    print(f"{key}={value}")
```

scientiac, 2025-06-16:
# On Niri

I haven't set anything, niri uses `wayland-1` by default.

```py
>>> import os
>>> display_wayland_var = os.environ.get('WAYLAND_DISPLAY')
>>> display_wayland_var
'wayland-1'
```

```py
>>> for key, value in os.environ.items():
...     print(f"{key}={value}")
...
SHELL=/usr/bin/zsh
GHOSTTY_BIN_DIR=/usr/bin
COLORTERM=truecolor
TERM_PROGRAM_VERSION=1.1.3-arch1
MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA=
DESKTOP_SESSION=niri
__ETC_PROFILE_NIX_SOURCED=1
XCURSOR_SIZE=24
XDG_SEAT=seat0
PWD=/home/scientiac
NIX_PROFILES=/nix/var/nix/profiles/default /home/scientiac/.nix-profile
XDG_SESSION_DESKTOP=niri
LOGNAME=scientiac
XDG_SESSION_TYPE=wayland
SYSTEMD_EXEC_PID=1444
_=/usr/bin/python
MOTD_SHOWN=pam
GDM_LANG=en_US.UTF-8
HOME=/home/scientiac
USERNAME=scientiac
LANG=en_US.UTF-8
XDG_CURRENT_DESKTOP=niri
MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/session.slice/niri.service/memory.pressure
WAYLAND_DISPLAY=wayland-1
NIX_SSL_CERT_FILE=/etc/ssl/certs/ca-certificates.crt
INVOCATION_ID=19d3dc83d73d432dae46ca86642c1b7d
NIRI_SOCKET=/run/user/1000/niri.wayland-1.1444.sock
MANAGERPID=1327
GHOSTTY_RESOURCES_DIR=/usr/share/ghostty
XDG_SESSION_CLASS=user
TERM=xterm-256color
TERMINFO=/usr/share/terminfo
USER=scientiac
GHOSTTY_SHELL_INTEGRATION_NO_SUDO=1
DISPLAY=:0
SHLVL=2
XDG_VTNR=2
XDG_SESSION_ID=3
XDG_RUNTIME_DIR=/run/user/1000
DEBUGINFOD_URLS=https://debuginfod.archlinux.org
JOURNAL_STREAM=9:18743
XCURSOR_THEME=Adwaita
XDG_DATA_DIRS=/home/scientiac/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/:/home/scientiac/.nix-profile/share:/nix/var/nix/profiles/default/share
PATH=/home/scientiac/.cargo/env:/home/scientiac/.local/bin:/home/scientiac/.nix-profile/bin:/nix/var/nix/profiles/default/bin:/home/scientiac/.cargo/bin:/home/scientiac/.local/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/scientiac/.spicetify
GDMSESSION=niri
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
MAIL=/var/spool/mail/scientiac
OLDPWD=/home/scientiac
TERM_PROGRAM=ghostty
KEYTIMEOUT=1
YSU_VERSION=1.10.0
STARSHIP_SHELL=zsh
STARSHIP_SESSION_KEY=1573312379296791
FZF_DEFAULT_COMMAND=fd --type f --hidden --follow --exclude .git
FZF_DEFAULT_OPTS=--height 40% --layout=reverse --border --inline-info
FZF_CTRL_T_COMMAND=fd --type f --hidden --follow --exclude .git
FZF_ALT_C_COMMAND=fd --type d --hidden --follow --exclude .git
EDITOR=nvim
VISUAL=nvim
PAGER=less
LESS=-R
QT_QPA_PLATFORM=wayland
QT_WAYLAND_DISABLE_WINDOWDECORATION=1
ELECTRON_OZONE_PLATFORM_HINT=wayland
SDL_VIDEODRIVER=wayland
GDK_BACKEND=wayland
MOZ_ENABLE_WAYLAND=1
CLUTTER_BACKEND=wayland
OZONE_PLATFORM=wayland
GTK_USE_PORTAL=1
```

# On GNOME

```py 
>>> import os
>>> display_wayland_var = os.environ.get('WAYLAND_DISPLAY')
>>> display_wayland_var
'wayland-0'
```

```py 
>>> for key, value in os.environ.items():
...     print(f"{key}={value}")
...
SHELL=/usr/bin/zsh
SESSION_MANAGER=local/einstein:@/tmp/.ICE-unix/12675,unix/einstein:/tmp/.ICE-unix/12675
GHOSTTY_BIN_DIR=/usr/bin
COLORTERM=truecolor
XDG_MENU_PREFIX=gnome-
TERM_PROGRAM_VERSION=1.1.3-arch1
GNOME_KEYRING_CONTROL=/run/user/1000/keyring
MEMORY_PRESSURE_WRITE=c29tZSAyMDAwMDAgMjAwMDAwMAA=
DESKTOP_SESSION=gnome
__ETC_PROFILE_NIX_SOURCED=1
PWD=/home/scientiac
NIX_PROFILES=/nix/var/nix/profiles/default /home/scientiac/.nix-profile
XDG_SESSION_DESKTOP=gnome
LOGNAME=scientiac
XDG_SESSION_TYPE=wayland
SYSTEMD_EXEC_PID=12830
XAUTHORITY=/run/user/1000/.mutter-Xwaylandauth.9K1372
MOTD_SHOWN=pam
GDM_LANG=en_US.UTF-8
HOME=/home/scientiac
USERNAME=scientiac
LANG=en_US.UTF-8
XDG_CURRENT_DESKTOP=GNOME
MEMORY_PRESSURE_WATCH=/sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/session.slice/org.gnome.SettingsDaemon.MediaKeys.service/memory.pressure
WAYLAND_DISPLAY=wayland-0
NIX_SSL_CERT_FILE=/etc/ssl/certs/ca-certificates.crt
INVOCATION_ID=4c7ad1691a644a0eb8d7a568e813b3e9
MANAGERPID=12554
GNOME_SETUP_DISPLAY=:1
GHOSTTY_RESOURCES_DIR=/usr/share/ghostty
XDG_SESSION_CLASS=user
TERM=xterm-256color
TERMINFO=/usr/share/terminfo
USER=scientiac
GHOSTTY_SHELL_INTEGRATION_NO_SUDO=1
DISPLAY=:0
SHLVL=1
XDG_RUNTIME_DIR=/run/user/1000
DEBUGINFOD_URLS=https://debuginfod.archlinux.org
JOURNAL_STREAM=9:89251
XDG_DATA_DIRS=/home/scientiac/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share/:/usr/share/:/home/scientiac/.nix-profile/share:/nix/var/nix/profiles/default/share
PATH=/home/scientiac/.cargo/env:/home/scientiac/.local/bin:/home/scientiac/.nix-profile/bin:/nix/var/nix/profiles/default/bin:/home/scientiac/.cargo/bin:/home/scientiac/.local/bin:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/scientiac/.spicetify
GDMSESSION=gnome
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
MAIL=/var/spool/mail/scientiac
GIO_LAUNCHED_DESKTOP_FILE_PID=13972
OLDPWD=/home/scientiac
TERM_PROGRAM=ghostty
_=/usr/bin/python
KEYTIMEOUT=1
YSU_VERSION=1.10.0
STARSHIP_SHELL=zsh
STARSHIP_SESSION_KEY=6178677313584208
FZF_DEFAULT_COMMAND=fd --type f --hidden --follow --exclude .git
FZF_DEFAULT_OPTS=--height 40% --layout=reverse --border --inline-info
FZF_CTRL_T_COMMAND=fd --type f --hidden --follow --exclude .git
FZF_ALT_C_COMMAND=fd --type d --hidden --follow --exclude .git
EDITOR=nvim
VISUAL=nvim
PAGER=less
LESS=-R
QT_QPA_PLATFORM=wayland
QT_WAYLAND_DISABLE_WINDOWDECORATION=1
ELECTRON_OZONE_PLATFORM_HINT=wayland
SDL_VIDEODRIVER=wayland
GDK_BACKEND=wayland
MOZ_ENABLE_WAYLAND=1
CLUTTER_BACKEND=wayland
OZONE_PLATFORM=wayland
GTK_USE_PORTAL=1
```

ldrahnik, 2025-06-16:
@scientiac Commited. Let's see if anyone will face an issue with this. I am concerned because it may cause problems, for example, during logging or on other distributions, but I have not found any associated historical issues. Can be reopened.

closed 2025-06-16 by ldrahnik