asus-linux-drivers/asus-numberpad-driver #246: [Bug]: Not working after installation - Ubuntu 25.04

mbootsman, 2025-07-05:
### Describe the bug

I did the install according to the docs. 
Laptop: Asus Zenbook UX3405M
Ubuntu 25.04
Attached the install log file.

No other log files are in this directory.

[install-05-07-2025-08-34-01.log](https://github.com/user-attachments/files/21072535/install-05-07-2025-08-34-01.log)

### Expected behavior

That holding the calculator icon on the Numberpad, showd the Numberpad.

### Relevant log output

```shell

```

### Desktop

- Program version [e.g. 57c742a]
- Distro [e.g. Arch Linux]
- Kernel [e.g. 5.18.15.arch1-2]

mbootsman, 2025-07-06:
Thanks @ldrahnik for the suggestion. I am using X11. I did a reinstall after modifying the files. The NumberPad is not working yet. Attached the install log again.

[install-06-07-2025-08-44-18.log](https://github.com/user-attachments/files/21085823/install-06-07-2025-08-44-18.log)

ldrahnik, 2025-07-06:
@mbootsman What does mean NumberPad is not working yet? I see no error. It should work. Can you provide runtime logs? When you create issue you can find instructions how to get them (`journalctl`).

ldrahnik, 2025-07-14:
Can be reopened.

closed 2025-07-14 by ldrahnik