asus-linux-drivers/asus-numberpad-driver #247: [Bug]: Some of the functionality in git does not work & AUR package does not work at all

Paradox-AT, 2025-07-22:
### Describe the bug

After installing AUR package [asus-numberpad-driver-g513-git](https://aur.archlinux.org/packages/asus-numberpad-driver-g513-git) the package is not working as intended.
The git installation works, but the functionality of the project is not working as intended.

Issues:
1) The systemd service is exiting because of a missing dependency: python-systemd.
2) After systemd service has started successfully the functionality of the project is not working at all i.e: num lock is not activated after long press on top right num lock symbol.
3) The git package when installed the activation works properly even for the calc but when pressing the buttons at a rapid pase the input gets cleared immediately and only the last pressed key remains as input.
4) The git package when installed without calc function still triggers calc when swiped from the left side
5) The brightness of the numpad can only be togged once i.e., from default brightness to max, it cannot be changed back to default or anything else

### Expected behavior

AUR installation should work properly and the intended functionality of the program should work

### Relevant log output

```shell

```

### Desktop

OS: Arch Linux x86_64
Host: ROG Strix G513QM_G513QM (1.0)
BIOS (UEFI): G513QM.331 (5.19)
Board: G513QM (1.0)
Kernel: Linux 6.15.6-arch1-1.1-g14
Init System: systemd 257.7-1-arch
DE: KDE Plasma 6.4.3
WM: KWin (Wayland)
Locale: en_IN.UTF-8
Mouse 1: ASUE1403:00 04F3:319A Mouse
Mouse 2: ASUE1403:00 04F3:319A Touchpad
Keyboard 1: Asus Keyboard
Keyboard 2: AT Translated Set 2 keyboard
TPM: TPM 2.0 Device

kamack38, 2025-07-22:
Hi. If you have an issue with the AUR package, I think it's better to report it in the AUR. I'm the maintainer of this package, so I didn't know it didn't work any more. If you find any solutions, you can create a PR to [this](https://github.com/kamack38/packages) repo.

Unfortunately, due to the way this driver functions, it's not easy to package it. I think you're better off installing it from source. Still, if you have any suggestions or improvements, I can happily discuss them with you.

ldrahnik, 2025-07-23:
@Paradox-AT Hello, thank you for creating an issue.

>The systemd service is exiting because of a missing dependency: python-systemd.

Should be fixed by [this](https://github.com/asus-linux-drivers/asus-numberpad-driver/commit/5e169ae0fdc03bbc699e778e44547a198deb6b15) commit.

>The git package when installed without calc function still triggers calc when swiped from the left side

Does this key still cause opening the calculator even without answering no to the question below during installation?

**Do you want try to install toggling script for XF86Calculator key? [y/N]?**

Because you may want to assign some other functionality to this icon, but then the icon has to send some key and XF86Calculator is appropriate I think. For me to not set up this key by default when answering no would be a good idea if any distribution automatically opens the calculator app for this key. What do you think?

Btw. it is not a problem to change it via `top_left_icon_slide_func_keys` from the layout file:
 
numberpad.py[541]:
```
top_left_icon_slide_func_keys = getattr(model_layout, "top_left_icon_slide_func_keys", [
    EV_KEY.KEY_CALC
])
```

If I am not mistaken, the rest should be on the AUR package @kamack38.

ldrahnik, 2025-07-25:
Can be reopened.

closed 2025-07-25 by ldrahnik