asus-linux-drivers/asus-numberpad-driver #254: [Bug]: Not working with fedora 43 (gnome 49)

rudrab, 2025-09-27:
### Describe the bug

Systemclt shows the unit is enabled  but starting failed.  So, the toggle switch in the mousepad is not enabling the numpad

### Expected behavior

touching the right top corner should enable the numpad

### Relevant log output

```shell
systemctl --user status asus_numberpad_driver@rudra.service
× asus_numberpad_driver@rudra.service - Asus NumberPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_numberpad_driver@.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/user/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Sat 2025-09-27 22:36:30 IST; 1min 16s ago
   Duration: 233ms
 Invocation: c5510d55a7824d95aaba8a7aae5365d8
    Process: 4694 ExecStart=/usr/share/asus-numberpad-driver/.env/bin/python3 /usr/share/asus-numberpad-driver/numberpad.py up5401ea /usr/s>
   Main PID: 4694 (code=exited, status=1/FAILURE)
   Mem peak: 37.5M
        CPU: 1.724s

Sep 27 22:36:30 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Scheduled restart job, restart counter is at 21.
Sep 27 22:36:30 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Start request repeated too quickly.
Sep 27 22:36:30 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Failed with result 'exit-code'.
Sep 27 22:36:30 roddur systemd[2599]: Failed to start asus_numberpad_driver@rudra.service - Asus NumberPad Driver.
Sep 27 22:36:30 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Consumed 1.724s CPU time, 37.5M memory peak.
Sep 27 22:37:07 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Start request repeated too quickly.
Sep 27 22:37:07 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Failed with result 'exit-code'.
Sep 27 22:37:07 roddur systemd[2599]: Failed to start asus_numberpad_driver@rudra.service - Asus NumberPad Driver.

journalctl --user -xeu asus_numberpad_driver@rudra.service

░░ Automatic restarting of the unit UNIT has been scheduled, as the result for
░░ the configured Restart= setting for the unit.
Sep 27 22:36:30 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Start request repeated too quickly.
Sep 27 22:36:30 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit UNIT has entered the 'failed' state with result 'exit-code'.
Sep 27 22:36:30 roddur systemd[2599]: Failed to start asus_numberpad_driver@rudra.service - Asus NumberPad Driver.
░░ Subject: A start job for unit UNIT has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit UNIT has finished with a failure.
░░ 
░░ The job identifier is 1256 and the job result is failed.
Sep 27 22:36:30 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Consumed 1.724s CPU time, 37.5M memory peak.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit UNIT completed and consumed the indicated resources.
Sep 27 22:37:07 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Start request repeated too quickly.
Sep 27 22:37:07 roddur systemd[2599]: asus_numberpad_driver@rudra.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ The unit UNIT has entered the 'failed' state with result 'exit-code'.
Sep 27 22:37:07 roddur systemd[2599]: Failed to start asus_numberpad_driver@rudra.service - Asus NumberPad Driver.
░░ Subject: A start job for unit UNIT has failed
░░ Defined-By: systemd
░░ Support: https://lists.freedesktop.org/mailman/listinfo/systemd-devel
░░ 
░░ A start job for unit UNIT has finished with a failure.
░░ 
░░ The job identifier is 1273 and the job result is failed.
```

### Desktop

I dont know the version, but its in sync with github repo.
I am in Fedora 43 (gnome 49)
```
uname -a
Linux roddur 6.17.0-0.rc7.56.fc43.x86_64 #1 SMP PREEMPT_DYNAMIC Mon Sep 22 14:29:30 UTC 2025 x86_64 GNU/Linux
```

ldrahnik, 2025-10-01:
@rudrab Thank you for creating an issue. What was the used python version? Try higher or equal to 3.10 or better try debug it like: 

# Debugging the ASUS NumberPad Driver with VS Code

1. **Install [Visual Studio Code](https://code.visualstudio.com/)** and the official **Python extension**.  
2. **Open the driver project folder** in VS Code.  
3. **Open** `numberpad.py`.  
4. In the **bottom-right status bar**, click the Python interpreter selector → **Browse** and point it to:  

   `/usr/share/asus-numberpad-driver/.env/bin/python3.10`  

   Use the version your environment provides (e.g. `python3.11`).  
   **Note:** Python ≥ 3.10 is required — `xcffib.xkb` is known to cause issues on **Python 3.9**. Version 3.10 is tested.  

5. With the Python extension, **set breakpoints** anywhere in the code.  
6. **Stop the systemd service** for the driver before debugging to avoid running two instances:  

   ```bash
   sudo systemctl stop <driver-service-name>

rudrab, 2025-10-01:
Thanks for your reply. I am using python3.14

ldrahnik, 2025-10-21:
@rudrab What is the output of `$ sudo i2cdetect -y 1`? And also of `... asus-numberpad-driver$ bash install_device_check.sh`?

rudrab, 2025-10-21:
Will it be safe to manually change the permission? I don't know what the file does.

rudrab, 2025-10-21:
@ldrahnik:

```
echo "i2c-dev" | sudo tee /etc/modules-load.d/i2c-dev-asus-numberpad-driver.conf
[sudo] password for rudra: 
i2c-dev
cat /etc/modules-load.d/i2c-dev-asus-numberpad-driver.conf 
i2c-dev
```

rudrab, 2025-10-21:
I have changed the permission:
```
 ll /dev/i2c-1
crw-rw----. 1 root root 89, 1 Oct 21 14:04 /dev/i2c-1
```

but numpad.py is still showing the same:
```
python numberpad.py up5401ea
2025-10-21 22:20:31,691 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2025-10-21 22:20:31,692 INFO Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2025-10-21 22:20:31,710 INFO Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
2025-10-21 22:20:31,711 INFO Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUP1415:00/0018:093A:300C.0001/input/input8
2025-10-21 22:20:31,711 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2025-10-21 22:20:31,711 ERROR Can't open the I2C bus connection (id: 1)
```

ldrahnik, 2025-10-21:
@rudrab `root root` this part is crucial, you need to see `root i2c` instead, linux group, try `chgrp`:

```
base) ldrahnik@Zenbook-UP5401EA:~/projects/asus-numberpad-driver$ sudo chgrp root /dev/i2c-1
[sudo] password for ldrahnik:          
(base) ldrahnik@Zenbook-UP5401EA:~/projects/asus-numberpad-driver$ ls -l /dev/i2c-1
crw-rw---- 1 root root 89, 1 Oct 21 18:55 /dev/i2c-1
(base) ldrahnik@Zenbook-UP5401EA:~/projects/asus-numberpad-driver$ sudo chgrp i2c /dev/i2c-1
(base) ldrahnik@Zenbook-UP5401EA:~/projects/asus-numberpad-driver$ ls -l /dev/i2c-1
crw-rw---- 1 root i2c 89, 1 Oct 21 18:55 /dev/i2c-1
(base) ldrahnik@Zenbook-UP5401EA:~/projects/asus-numberpad-driver$ 
```

ldrahnik, 2025-10-22:
@rudrab Can you for example using `pyenv` (in Readme.MD is described how) or using `conda` environment try python version 3.10? 

```
conda create -n i2c python=3.10 smbus2
conda activate i2c
python
... and again the same
Python 3.10.18 (main, Sep  3 2025, 20:02:44) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
Ctrl click to launch VS Code Native REPL
>>> from smbus2 import SMBus
>>> bus = SMBus()
>>> bus.open(1)
>>>
```

So we could confirm / exclude following:

> That SystemError: buffer overflow when calling bus.open(1) in Python 3.14 comes from an internal incompatibility between the smbus2 package and the new Python C API changes in 3.13+.

s-badran, 2025-10-26:
@rudrab at your convenience time, pls give this [fix](https://github.com/s-badran/asus-numberpad-driver/tree/fix-i2c-permission-issue) a try and share your feedback

rudrab, 2025-11-01:
changed the permission:
```
ll i2c-1
crw-rw----. 1 root i2c 89, 1 Nov  1 09:31 i2c-1
```
But, still:
```
 systemctl start --user asus_numberpad_driver@rudra.service 
Job for asus_numberpad_driver@rudra.service failed because the control process exited with error code.
See "systemctl --user status asus_numberpad_driver@rudra.service" and "journalctl --user -xeu asus_numberpad_driver@rudra.service" for details.
```

ldrahnik, 2025-11-01:
@rudrab And output of:

```
$ pip install python-periphery

$ python
Python 3.10.14 | packaged by conda-forge | (main, Mar 20 2024, 12:45:18) [GCC 12.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
Ctrl click to launch VS Code Native REPL
>>> from periphery import I2C
>>> path = f"/dev/i2c-1"
>>> i2c = I2C(path)
>>> i2c.close()
```

rudrab, 2025-11-01:
```
>>> i2c = I2C(path)
# before changing the permission
Traceback (most recent call last):
  File "/home/rudra/.local/lib/python3.14/site-packages/periphery/i2c.py", line 72, in _open
    self._fd = os.open(devpath, os.O_RDWR)
               ~~~~~~~^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<python-input-3>", line 1, in <module>
    i2c = I2C(path)
  File "/home/rudra/.local/lib/python3.14/site-packages/periphery/i2c.py", line 58, in __init__
    self._open(devpath)
    ~~~~~~~~~~^^^^^^^^^
  File "/home/rudra/.local/lib/python3.14/site-packages/periphery/i2c.py", line 74, in _open
    raise I2CError(e.errno, "Opening I2C device: " + e.strerror)
periphery.i2c.I2CError: [Errno 13] Opening I2C device: Permission denied

```
# after changing permission and group
```
>>> i2c = I2C(path)
Traceback (most recent call last):
  File "<python-input-4>", line 1, in <module>
    i2c = I2C(path)
  File "/home/rudra/.local/lib/python3.14/site-packages/periphery/i2c.py", line 58, in __init__
    self._open(devpath)
    ~~~~~~~~~~^^^^^^^^^
  File "/home/rudra/.local/lib/python3.14/site-packages/periphery/i2c.py", line 81, in _open
    fcntl.ioctl(self._fd, I2C._I2C_IOC_FUNCS, buf, True)
    ~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
SystemError: buffer overflow
>>> 
```

ldrahnik, 2025-11-02:
@rudrab Try the latest master branch now, please.

rudrab, 2025-11-02:
Still the same error. With python 3.14, find some more problem:

`pyinotify` module has changed:
```
./numberpad.py 
Traceback (most recent call last):
  File "/home/rudra/Utility/asus-numberpad-driver/./numberpad.py", line 17, in <module>
    from pyinotify import WatchManager, IN_CLOSE_WRITE, IN_IGNORED, IN_MOVED_TO, AsyncNotifier, ProcessEvent
ImportError: cannot import name 'AsyncNotifier' from 'pyinotify' (/usr/lib/python3.14/site-packages/pyinotify.py). Did you mean: 'AsyncioNotifier'?
```

```
./numberpad.py up5401ea
2025-11-02 19:48:39,449 INFO Detecting keyboard from string: "N: Name="AT Translated Set 2 keyboard""
2025-11-02 19:48:39,449 INFO Set keyboard 3 from H: Handlers=sysrq kbd leds event3
2025-11-02 19:48:39,463 INFO Detecting touchpad from string: "N: Name="ASUP1415:00 093A:300C Touchpad""
2025-11-02 19:48:39,463 INFO Set touchpad device id 1 from S: Sysfs=/devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-1/i2c-ASUP1415:00/0018:093A:300C.0001/input/input8
2025-11-02 19:48:39,463 INFO Set touchpad id 5 from H: Handlers=mouse1 event5
2025-11-02 19:48:39,464 INFO Touchpad min-max: x 0-3996, y 0-2242
2025-11-02 19:48:39,464 INFO Numpad min-max: x 200-3796, y 200-2162
Traceback (most recent call last):
  File "/usr/lib64/python3.14/site-packages/pywayland/dispatcher.py", line 46, in dispatcher_func
    ret = func(self, *args)
  File "/home/rudra/Utility/asus-numberpad-driver/./numberpad.py", line 604, in wl_keyboard_keymap_handler
    wl_load_keymap_state()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/home/rudra/Utility/asus-numberpad-driver/./numberpad.py", line 576, in wl_load_keymap_state
    load_evdev_key_for_wayland(char, keyboard_state)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rudra/Utility/asus-numberpad-driver/./numberpad.py", line 521, in load_evdev_key_for_wayland
    mod_masks_for_level = keymap.key_get_mods_for_level(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Keymap' object has no attribute 'key_get_mods_for_level'. Did you mean: 'key_get_syms_by_level'?
^C2025-11-02 19:48:45,791 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/home/rudra/Utility/asus-numberpad-driver/./numberpad.py", line 2927, in <module>
    sleep(0.5)
    ~~~~~^^^^^
KeyboardInterrupt
2025-11-02 19:48:45,792 INFO Clean up started
2025-11-02 19:48:45,800 INFO Clean up finished
2025-11-02 19:48:45,800 INFO Exiting

```

ldrahnik, 2025-12-06:
@rudrab Could you please test whether the latest master fixed problem with not automatically setting up `uinput` and `i2c` group via udev rules? Providing outputs of commands `$ udevadm info --attribute-walk --name=/dev/i2c-0`  and `$ udevadm info --attribute-walk --name=/dev/uinput` would be helpful if not. Thank you.